vectorgrp/sil-kit #2: Need help to establish communication between multiple ECU's

AasaiAlangaram, 2023-03-07:
**Describe the issue**
Hi. I'm trying to establish connection between two venvironment created using Yaml. However, I couldn't find any demo or reference to do the same. Initially, tried the CAN Demo which comes as part of Vector SIL kit package. That worked because in which communication established between one SUT(CAN writer) and .venvrionment(CAPL responding to CAN Writer req using on msg). 

Wondering, how to do the same with two venvrionment.

Command used to create venvironment: 
1.environment-make ECU1.yaml
2.environment-make ECU2.yaml

**To Reproduce**
Steps to reproduce the behavior:
1. Created two yaml for two ECU's
2. Generated .venvironment using environment-make command

**Expected behavior**
Establish communication between two ecu's which are part of .venvironment created using yaml.

**Your Environment (please complete the following information):**
 - OS version and platform: Windows 10

NilsNwr, 2023-03-08:
Dear AasaiAlangaram,
thank you for filing this issue.
Since your question relates to venvironment, please contact the official Vector Support at support.vector.com for support to your issue.

AasaiAlangaram, 2023-03-08:
Okay @NilsNwr. Thanks for letting me know. :)

closed 2023-03-08 by AasaiAlangaram