vectorgrp/sil-kit #3: [Request] Add support for VS2019

aurbain-avs, 2023-03-31:
In order to integrate VSK into my build system, i need support of the Visual Studio 2019 compiler.

aurbain-avs, 2023-03-31:
I made the preset and i fixed some TU.
So i can make a PR for this issue.

MariusBgm, 2023-03-31:
Thanks. Currently we don't take external contributions, but we are working on this!
According to Microsoft documentation, the VS2017 should work in your setup, as long as you link all object files using the VS2019 linker: see https://learn.microsoft.com/en-us/cpp/porting/binary-compat-2015-2017?view=msvc-170
Quote from there:
> You can mix binaries built by different versions of the v140, v141, v142, and v143 toolsets. However, you must link by using a toolset at least as recent as the most recent binary in your app

aurbain-avs, 2023-03-31:
We also need to link with lot of other libraries that VSK has as third parties and we already have in our build system.
So it's why I think it's preferable to add the support for all targets i've mentionned in the issues i have created.
I'd prefer that they are natively supported by VSK instead to we have to rebuild all for each new version we want to use.
But we can also work by building all by us.

NilsNwr, 2023-03-31:
Just to clarify: You should be able to use our VS2017 packages in your VS2019 pipeline. 
You should be able to use the Ubuntu 18.04 packages in Ubuntu 20.04.
You are probably able to use the Ubuntu 18.04 packages in you CentOS setup.

Have you tried this before you stick to building the SIL Kit packages yourself?

Sidefact: VSK is not an official abbreviation for Vector SIL Kit. We always reccomend using "SIL Kit" to prevent misunderstandings.