vectorgrp/sil-kit #5: [Request]Add support for CentOS 7 - GCC 8

aurbain-avs, 2023-03-31:
Add support for CentOS 7 and GCC 8.

GCC 8 is avaible on CentOS 7 by installing the Software Collections repository:
`yum install centos-release-scl`

Then the GCC 8 compiler and tools suite can be installed:
`yum install devtoolset-8-gcc devtoolset-8-gcc-c++`

MariusBgm, 2023-03-31:
We currently don't plan to provide binary packages for this platform.
You can easily build this on your own.
We have a cmake preset for gcc8 (gcc8-release) which you can adapt to suit your needs (see CMakePresets.json, and `cmake --list-presets`).
The CMake `package` target will generate a release zip file for you.
For example:
```
cmake --preset gcc8-release
cmake --build --preset gcc8-release
cmake --build --preset gcc8-release --target package
```
Having said that, the Ubuntu18.04 binary packages might work here too, as we build with gcc-8 and have no dependencies other than glibc.

aurbain-avs, 2023-03-31:
OK, i'll look at it.

closed 2023-05-02 by MariusBgm