vectorgrp/sil-kit #7: QuickStart not running

jesko42, 2023-05-22:
**Describe the issue**
I tried the plain QuickStart example and it did not run.

**To Reproduce**
Steps to reproduce the behavior:
1. [SIL-KIT is already installad and runnable]
2. Go to 'https://vectorgrp.github.io/sil-kit-docs/usage/quickstart.html'
3. Download the three files [CMakeLists.txt, simple.cpp, simple.yaml]
4. Copy to new directory
5. go to dir
6. "mkdir build"
7. "cd build"
8. "cmake .."
9. "cmake --build ."
10. "cp ../simple.yaml ."
11. open another two terminals [#2, #3]
12. in the terminal #2 enter command: "sil-kit-registry"
13. in the terminal #3 enter command: "sil-kit-system-controller PublisherParticipant subscriberParticipant"
14. in the first terminal (#1) enter: "./SampleSilKit"

**Expected behavior**
Some output to see. Nothing happens. Looks like the lambda function producing the data is not called ... Simulation may not start ...

**Screenshots**

**Your Environment (please complete the following information):**
 - OS version and platform: Linux, Ubuntu 22.04.2 LTS
 - Compiler and CMake version: gcc (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0, cmake version 3.26.3
 - SIL Kit Version: IL Kit version: 4.0.24

**Additional context**
nope

jesko42, 2023-05-22:
OK, just after writing the issue I saw, that there was a typo in the documentation. Instead of
"sil-kit-system-controller PublisherParticipant SubscriberParticipant" it was written
"sil-kit-system-controller PublisherParticipant **s**ubscriberParticipant"

Maybe Windows does not distinguish, but with Linux ...

There is another neglegtable typo in the first main()-fragment:
```
int main(int argc, char** argv)
{
    auto config = SilKit::Config::ParticipantConfigurationFromFile("simple.yaml")
    // TODO: Use config to create participants
}
```

The 3rd line must be: `auto config = SilKit::Config::ParticipantConfigurationFromFile("simple.yaml")**;**`

MariusBgm, 2023-05-22:
@jesko42  thanks for pointing out the missing `;` in the main()-fragment. We will fix that ASAP. However, I can't find the misspelled `subscriberParticipant` with lower-case `s` in the linked document. By the way, the participant names given to the `sil-kit-systemcontroller` are indeed case-sensitive, independent of the platform you run your SIL Kit application.

jesko42, 2023-05-22:
@MariusBgm tanks for keeping an eye on it, unfortunately I'm a little late ...
You are right! If I *now* look at it everything seems to be fine - I don't know how this could happen !?
Now you know: It's working, as I have tested it here :-) :+

closed 2023-05-23 by MariusBgm