vectorgrp/sil-kit #9: program crashes without apparent reason when instantiating SilKit::Config::ParticipantConfigurationFromFile object

fskoras, 2023-12-13:
**Describe the issue**
I was trying to compile a simple example program from provided documentation "Developer Guide" chapter.
I am using vscode with cmake extension configuring visual studio 17 2022 toolchain

**To Reproduce**
follow the tutorial from documentation in specified environment

**Expected behavior**
A clear and concise description of what you expected to happen.
I expect executable not to crash

**Screenshots**

CMakeLists.txt
---
[image]

simple.cpp
---
[image]

GDB output
---
[image]

**Your Environment (please complete the following information):**
 - OS version: Windows 10
 - Compiler and CMake version: visual studio 17 2022, cmake version 3.28.0
 - SIL Kit Version: 4.0.43
 - Please include a description of the SIL Kit participants, including which
   controllers are created and the configuration applied to them: N/A

**Additional context**

When I try to compile the program with `config` object the program crash right after startup.
That means I am not even able to create a breakpoint in line 14 to see whats going on.

MariusBgm, 2023-12-14:
Hi @fskoras , the Windows return code 0xC0000134 means DLL not found (STATUS_DLL_NOT_FOUND). When you run your sample exeucutable, make sure that you have SilKit.dll in the same folder, or the path containing it is in your PATH variable. That is not a crash in SilKit, it's Windows unable to load the dependencies. Hope that helps

fskoras, 2023-12-14:
You are correct, that was it. I was just surprised I didn't receive any notification from the operating system that the dll failed to load. thank you for your help

MariusBgm, 2023-12-14:
Great that it works for you. 
For what it's worth, If you start the executable in an explorer window you should get a pop-up dialog with an error message, if you run it in a shell or debugger, it doesn't show...

closed 2023-12-14 by MariusBgm