vectorgrp/sil-kit #10: Build with statically linked libs?

Andoramb, 2023-12-19:
**Describe the issue**
Is it possible to build (for example) the sil-kit-registry binary with statically linked libraries, so it can run standalone on different environments?

**To Reproduce**
So far I have added the following lines to the `cmake` instruction:
```
-D CMAKE_FIND_LIBRARY_SUFFIXES=".a" \
-D BUILD_SHARED_LIBS=OFF \
```

**Expected behavior**
When I try to run the compiled binary on an other OS (same arch as origin) it still gives missing library related errors:
```
./sil-kit-registry: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by ./sil-kit-registry)
./sil-kit-registry: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by ./sil-kit-registry)
./sil-kit-registry: /lib/x86_64-linux-gnu/libstdc++.so.6: version `CXXABI_1.3.13' not found (required by ./sil-kit-registry)
./sil-kit-registry: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by ./sil-kit-registry)
./sil-kit-registry: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by ./sil-kit-registry)
./sil-kit-registry: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by ./sil-kit-registry)
```

**Your Environment (please complete the following information):**
 - OS version and platform: 
   *same for both OS-es
   From `uname -a`:  
   `Linux DESKTOP microsoft-standard-WSL2 #1 x86_64 x86_64 x86_64 GNU/Linux`
 - Compiler and CMake version:
     Overall the whole `cmake` command looks like this:
```
cmake -D SILKIT_BUILD_DOCS=OFF \
      -D SILKIT_BUILD_TESTS=OFF \
      -D SILKIT_BUILD_DEMOS=OFF \
      -D SILKIT_BUILD_UTILITIES=ON \
      -D CMAKE_INSTALL_PREFIX=/sil-kit \
      -D CMAKE_FIND_LIBRARY_SUFFIXES=".a" \
      -D BUILD_SHARED_LIBS=OFF \
      .. 
cmake --build . -j 4
cmake --build . --target install
```
 - SIL Kit Version:
   Doesn't really matter
 - Please include a description of the SIL Kit participants, including which
   controllers are created and the configuration applied to them:
   N/A atm

VDanielEdwards, 2023-12-19:
Hello @Andoramb, we do not support linking SIL Kit as a static library, and neither do we support statically linking the utility executables.

Thank you for your message and happy holidays!

Andoramb, 2023-12-21:
Alright,

Good to know!
Thanks for the quick answer.
Same to you 👍

closed 2023-12-21 by Andoramb