vectorgrp/sil-kit #49: No deterministic simulation even though time synchronization service is used

VladislavNeufeld, 2024-05-26:
Hello,

I am using your benchmark demo in order to investigate whether the sil kit has the ability to work deterministic. For that I have extended the demo with a logging function that is called after each message transmit and saves the participant ID.

I have configured just a very simple simulation using two participants:
SilKitDemoBenchmark.exe 1 1 2 1 69 --configuration ./SilKit-Demos/Benchmark/DemoBenchmarkDomainSocketsOff.silkit.yaml

I am observing that each each simulation run (each execution of the demo) produces a different logging sequence, what means that the participants are not sending the messages in the same order during each simulation run.

Is there anything I can do to ensure that the sequence of events is equal for each simulation run? Or does the Silkit not support sequence determinism?

BR

VladislavNeufeld, 2024-06-01:
Hello @KonradBkd ,

thank you for clarification!
I just want to make sure that the frame event callback is finished before the next simulation step is executed (before next message is sent). In our current solution (other than silkit) it is also possible that frame event callbacks are processed downstream. This makes it sometimes tough to reproduce error cases. That's why we currently considering to use the SIL Kit and why I am explorate the capabilities of the SIL Kit.

I am thinking how I can verify your statement regarding the determinism guarantee of SIL kit. What do you think about the following approach:
- add logging to the benchmarking demo for each participant (each participant has his own log file)
- logging of message reception callback
- logging of simulation step handler event

Then I can check in the logs of each participant that there is always the same sequence of events, what means that all messages are received before the next simulation step starts. For example if messages per simulation step is configured to be 2, then I would have the following sequence:
"SimulationStep"
"MessageReception"
"MessageReception"
"SimulationStep"
"MessageReception"
"MessageReception"
...
...

Can I verify it like that or do you see any pitfall with this approach?
I have tried it out and have encountered some sequence violations. That means that I have received messages from the past.

KonradBkd, 2024-06-03:
> I just want to make sure that the frame event callback is finished before the next simulation step is executed (before next message is sent)

One thing to know here is that the handlers in the SIL Kit are all invoked from the same thread (called "IO \<participant name\>"), so you will never get concurrent calls of SIL Kit handlers. The great advantage here is that you don't have to take care of protecting your data from parallel access in the handlers. Also debugging is more easy, as all handlers appear sequentially.

Your approach to log in the reception- and simulation step handlers makes sense, but you have to be a bit careful with the interpretation. Basically, you need to look at the timestamps as well. In the `SimulationStepHandler`, you can validate against the guarantee: 

> "_When the SimulationStepHandler is called for time t_sim, all messages with t_msg < t_sim (from the past) have been received._". 

On the reception side, the equivalent guarantee is: 

> "_Incoming messages have a sending timestamp t_msg  >= t_sim (from the future)_".

I quickly adapted the BenchmarkDemo and got the sequence:

```
[Participant0] [info] SimStep t = 0.000000ms
[Participant0] [info] receive data sent at t = 0.000000ms
[Participant0] [info] receive data sent at t = 0.000000ms
[Participant0] [info] SimStep t = 1.000000ms
[Participant0] [info] receive data sent at t = 1.000000ms
[Participant0] [info] receive data sent at t = 1.000000ms
[Participant0] [info] SimStep t = 2.000000ms
[Participant0] [info] receive data sent at t = 2.000000ms
[Participant0] [info] receive data sent at t = 2.000000ms
...
[Participant0] [info] SimStep t = 203.000000ms
[Participant0] [info] receive data sent at t = 203.000000ms
[Participant0] [info] receive data sent at t = 203.000000ms
[Participant0] [info] receive data sent at t = 204.000000ms // <- Can happen, but does not break the guarantee
[Participant0] [info] SimStep t = 204.000000ms
[Participant0] [info] receive data sent at t = 204.000000ms
[Participant0] [info] SimStep t = 205.000000ms
[Participant0] [info] receive data sent at t = 205.000000ms
[Participant0] [info] receive data sent at t = 205.000000ms
```

Probably, you also saw such a "hickup" in your logs which breaks the nice sequence but not the guarantee.

To reproduce:
- Add a `auto* logger = participant->GetLogger();` and capture all by recerence in handlers (`[&]`) for convenience. 
- Log the simulation time in the `SimulationStepHandler`: `logger->Info("SimStep t = " + std::to_string(now.count() * 1e-6) + "ms");`
- Log the timestamp of the incoming message in the reception handler. Additionally, you can add a violation check for the error case `data.timestamp < timeSyncService->Now()` which should never happen: 
```
    participant->CreateDataSubscriber(
        "SubCtrl1", matchingDataSpec, [&](auto*, const DataMessageEvent& data) {
        // this is handled in I/O thread, so no data races on counter.
        messageCounter++;
	logger->Info("receive data sent at t = " + std::to_string(data.timestamp.count() * 1e-6) + "ms");
        if (data.timestamp < timeSyncService->Now()) // Violation check
        {
            logger->Error("Chronology error!");
        }
    });
```

As mentioned, one pitfall here is that you must send data only in the `SimulationStepHandler`. Feel free to show some code if you have trouble reproducing this behavior.

VladislavNeufeld, 2024-06-04:
Thank you for explanation, now it is clear. That means some participants are scheduled earlier by the OS and have sometimes already executed the simulation step handler and others not and therefore the "hickup".

I was able to reproduce it. Another thing that I have observed is that sometimes for some participants the last simulation step handler call is not executed/missing in the log sequence.

For example with this call: SilKitDemoBenchmark 1 1 10 10 10 --configuration logging.silkit.yaml

for some participants sometimes "SimStep t = 1001" can not be found. 

How can this be explained? How can it happen that some participants are not executing the last step?

Now I am even confused how the system can reach this step since I have set the simulation duration to 1 second virtual time (1 ms step size). So I would expect that SimStep t = 1000 would be the last step.

KonradBkd, 2024-06-05:
You're welcome! The benchmark demo implements the stop like:
```
if (now > benchmark.simulationDuration)
{
    lifecycleService->Stop("Simulation done");
}
```
Actually, only a single participant would need to call this and the others would stop as well. This is the behavior of the `OperationMode::Coordinated` specified in `CreateLifecycleService`. For simplicity of the Demo, every participant just calls `Stop()`.

Now it can happen that participant A has reached 1001ms (`now > benchmark.simulationDuration`), calls `Stop()` and finishes his SimulationStep. Another participant now might receive the Stop-Signal from A _before_ of _after_ he starts his step for 1001ms. Due to the concurrency of the SimulationSteps we can't really control this.
To get a safe limit for the simulation time, you can guard your simulation logic with respect to time:

```
if (now > 1000ms) 
{
    lifecycleService->Stop("Simulation done"); // Triggers at 1001ms but some participants might not reach this!
}
else
{
    Publish(...) // Guaranteed to trigger everywhere in the interval [0ms, 1000ms]!
}
```

VladislavNeufeld, 2024-06-07:
Thank you for the support!

closed 2024-06-07 by VladislavNeufeld