vectorgrp/sil-kit #54: Extended Id j1939 messages not working with sil-kit on canoe

RodrigoBaumann-hub, 2024-06-10:
Hi, I'm trying to communicate J1939 messages with SIL-KIT, but every time I send a message, the ID gets truncated. For example: 
vcan2  400   [8]  C0 00 00 00 00 F0 FF 00
Also, when I send a message from WSL to Canoe with an extended ID like this:
cansend vcan2 18FDE5FF#13140003FF00660101

I get this error:

[E]	[System]	83-0203 CAN frame with invalid format received on channel CAN 7.

What could the solution be? Is it an issue with Canoe's configuration or SIL-KIT?

VDanielEdwards, 2024-06-11:
Hello @RodrigoBaumann-hub, thank you for your message!

I'm investigating your issue!

VDanielEdwards, 2024-06-11:
Hello @RodrigoBaumann-hub, thanks to your report we have found a bug in the `sil-kit-adapters-vcan` project.

I created a [PR in the `sil-kit-adapters-vcan` repository](https://github.com/vectorgrp/sil-kit-adapters-vcan/pull/2) that addresses the issue that CANoe received invalid CAN frames and aborts the measurement.

VDanielEdwards, 2024-06-11:
@RodrigoBaumann-hub, regarding the other issue, where CAN IDs get truncated: could you provide us with the 'original' frame, and the 'transformed' frame with the truncated ID?

VDanielEdwards, 2024-06-11:
@RodrigoBaumann-hub, we believe that we have found the other problem as well. Please don't be confused if the PR on the `sil-kit-adapters-vcan` repository is closed without merging - the issues are handled internally!

RodrigoBaumann-hub, 2024-06-11:
Thank you @VDanielEdwards 
If I send 0x18FECC00 AA BB CC DD EE FF 11 22 via Canoe and receive it on WSL, I get:
vcan2 400 [8] AA BB CC DD EE FF 11 22
However, the ID is truncated. I'm working with a network on Canoe that sends J1939 messages and also forwards them via SIL-KIT to vcan2 on WSL.

VSvenMorgenthaler, 2024-06-13:
@RodrigoBaumann-hub (CC: @VDanielEdwards): The issue has been solved on: https://github.com/vectorgrp/sil-kit-adapters-vcan/commit/65eec259d5c159b318dc54ba5ee6e97c9fb5ba3b

VDanielEdwards, 2024-06-28:
Closing this issue due to inactivity. Please feel free to reopen this, if required.

closed 2024-06-28 by VDanielEdwards