vectorgrp/sil-kit #65: Performance concerns

fskoras, 2024-06-27:
Hello,

I have prepared a simple simulation consisting of 3 participants (one of them is system controller)
Each participant starts on a separate thread

I was trying to run an empty simulation on local host for duration 60s in Debug mode and it took about 17s real time (I didn't even use gdb).
Is this performance expected?
Could you please advise?

Kind Regards,
Filip

Simulation spec
---
SIL Kit version: 4.0.50
operation mode: coordinated
number of participants: 3
services used: lifecycle, time-sync
initialStepSize: 1ms
SetSimulationStepHandler: empty lambda function (for each participant)

System spec
---
Processor	11th Gen Intel(R) Core(TM) i7-11850H @ 2.50GHz, 2496 Mhz, 8 Core(s), 16 Logical Processor(s)
OS Name	Microsoft Windows 10 Enterprise
Version	10.0.19045 Build 19045
Installed Physical Memory (RAM)	32,0 GB

Files
---
[CMakeLists.txt](https://github.com/user-attachments/files/16014826/CMakeLists.txt)
[simple.cpp.txt](https://github.com/user-attachments/files/16014829/simple.cpp.txt)
[simple.yaml.txt](https://github.com/user-attachments/files/16014830/simple.yaml.txt)

VDanielEdwards, 2024-06-27:
Hello Filip,

thank you for your message! We think that since you are setting the `CMAKE_BUILD_TYPE` inside of your `CMakeLists.txt`, this causes your executable to always link against the debug-build of SIL Kit.

The pre-built SIL Kit packages not only contain the release-build of SIL Kit (`SilKit.dll` / `libSilKit.so`), but additionally a debug-build called `SilKitd.dll` (`libSilKitd.so`).

Could you check which `.dll` (or `.so`) file is linked into your executable?

When building with `CMAKE_BUILD_TYPE` set to `Debug`, you should get the following dependencies:

```
> dumpbin.exe /dependents .\SampleSilKit.exe
[...]
  Image has the following dependencies:

    SilKitd.dll
[...]
```

While building with `CMAKE_BUILD_TYPE` set to `Release`, you should find this:

```
> dumpbin.exe /dependents .\SampleSilKit.exe
[...]
  Image has the following dependencies:

    SilKit.dll
[...]
```

On Linux you can use `ldd` for the same purpose (`ldd SampleSilKit`).

The pre-built package contains both Debug and Release for historical reasons. We are however currently considering changing the package structure, such that we only delivery a single, optimized binary.

Kind regards,
Daniel

fskoras, 2024-06-28:
You are indeed correct, my binary executable was linking with `SilKitd.dll`

So do I have to explicitly specify link properties for my target to force CMake to link with release `SilKit.dll` like this?

```cmake
cmake_minimum_required(VERSION 3.28)
Project(SampleSilKit)

set (CMAKE_CXX_STANDARD 20)
find_package(Threads REQUIRED)
add_executable(${PROJECT_NAME} simple.cpp)
target_link_directories(${PROJECT_NAME} PRIVATE C:/Tools/SilKit/current/SilKit/lib)
target_link_libraries(${PROJECT_NAME} PRIVATE SilKit Threads::Threads)
target_include_directories(${PROJECT_NAME} PRIVATE C:/Tools/SilKit/current/SilKit/include)

```

Or is it perhaps possible to do with `find_package`?
I want all my other build binaries to be built with `Debug` configuration

VDanielEdwards, 2024-06-28:
Thank you for confirming! This is a bit of a hack, but for now you should be able to override the selected import-library and location properties for the `SilKit::SilKit` target like so:

```cmake
find_package(SilKit 
    REQUIRED CONFIG
    PATHS "C:/Tools/SilKit-4.0.43-Win-x86_64-VS2017"
)

get_target_property(_SILKIT_IMPORTED_IMPLIB_RELEASE SilKit::SilKit IMPORTED_IMPLIB_RELEASE)
get_target_property(_SILKIT_IMPORTED_LOCATION_RELEASE SilKit::SilKit IMPORTED_LOCATION_RELEASE)

set_target_properties(SilKit::SilKit PROPERTIES
    IMPORTED_IMPLIB_DEBUG "${_SILKIT_IMPORTED_IMPLIB_RELEASE}"
    IMPORTED_LOCATION_DEBUG "${_SILKIT_IMPORTED_LOCATION_RELEASE}"
)
```

fskoras, 2024-06-28:
I just tried it right away and it works perfectly!

Your suggestion is a better approach than mine because it allows to inherit SilKit library properties through simple `target_link_libraries` call which is exactly what I wanted.

Thank you

VDanielEdwards, 2024-06-28:
You're welcome! If you encounter any other issue, please feel free to create a new issue! Have a good weekend!

closed 2024-06-28 by VDanielEdwards