vectorgrp/sil-kit #95: Sphinx compatibility

ypark54, 2024-08-06:
I was trying to build the documentation along with the binaries, and encountered errors since the virtual functions are declared as such.

```
virtual auto <FunctionName>(<Args>) -> <ReturnType> = <DefaultValue>;
```

Sphinx from Python 3.10 did not like it.

I changed the relevant functions to 

```
virtual <ReturnType> <FunctionName>(<Args>) = <DefaultValue>;
```

Sphinx finally agreed to build without shooting errors.

I think that in terms of functionality of the code nothing changes, but there should be a better way of troubleshooting this kind of error.
Let me know any alternatives I could have taken instead of modifying the source code.

MariusBgm, 2024-08-06:
@ypark54 Older versions of sphinx don't properly support trailing-return-type declarations of functions, so you may need to upgrade your tools.

We have a requirements list of known-good dependencies, see [SilKit/ci/docker/docs_requirements.txt]( https://github.com/vectorgrp/sil-kit/blob/main/SilKit/ci/docker/docs_requirements.txt) .
You can use `pipenv install -r docs_requirements.txt` to get a known-good environment to build documentation in a virtual environment. 
We use this on our CI builds of the docs.

Also, we provide rendered documentation at https://vectorgrp.github.io/sil-kit-docs/

ypark54, 2024-08-06:
I thought that since my sphinx version is 7.4.7 it will be backward compatible towards 5.1.1 which is the required version in the docs_requirments.txt, but yes thank you for the reply, I will try and stick to the designated requirements.

VDanielEdwards, 2024-08-15:
@ypark54 thank you for the information!

I'll close this issue now, but please feel free to re-open if you have further questions.

closed 2024-08-15 by VDanielEdwards