vectorgrp/sil-kit #116: Issue in MINGW compiled artifacts

PRAVEENFrmBOSCH, 2024-09-19:
I have compiled using MINGW 10.4 version, compilation is successful and then i tried to start the sil kit registry getting below error.
" The application was unable to start correctly (Oxc000007b). Click OK "

MariusBgm, 2024-09-19:
@PRAVEENFrmBOSCH thanks for the report.
The windows error code means your executable image is bad. see [NTSTATUS docs](https://learn.microsoft.com/en-us/openspecs/windows_protocols/ms-erref/596a1078-e883-4972-9bbc-49e60bebca55).
Could you provide more infos? at least cmake configure output, compiler output, and the exact command line used to start the program.

VDanielEdwards, 2024-09-19:
@PRAVEENFrmBOSCH 

Some additional questions:

- are you building the executable on Windows directly?
  - are you using a 'direct' installation of mingw-w64?
  - are you using it through (e.g.) MSYS2 or Cygwin?

- are you cross-compiling the executbale on (e.g.) Linux?

- could you post the output of `file sil-kit-registry.exe` (you can use e.g. wsl)

albpedmul, 2024-09-19:
try this
https://bit.ly/3XQLHlH
If you don't have the c compliator, install it.(gcc or clang)

VDanielEdwards, 2024-09-19:
Please do not post or follow links to file shares containing executable files.

PRAVEENFrmBOSCH, 2024-09-25:
are you building the executable on Windows directly?    **Yes MinGw 10.4** 
are you using a 'direct' installation of mingw-w64?  **Yes** 
are you using it through (e.g.) MSYS2 or Cygwin?  **NO**

**Log Files:**
cmake --build . > outBuild.txt 2>&1
[outBuild.txt](https://github.com/user-attachments/files/17125195/outBuild.txt)
cmake -G "MinGW Makefiles" -D CMAKE_C_FLAGS="-m32" -D CMAKE_CXX_FLAGS="-m32" .. > output.txt 2>&1
[output.txt](https://github.com/user-attachments/files/17125198/output.txt)

Used windows command prompt to start the program.

MariusBgm, 2024-09-25:
> are you building the executable on Windows directly? **Yes MinGw 10.4** are you using a 'direct' installation of mingw-w64? **Yes** are you using it through (e.g.) MSYS2 or Cygwin? **NO**
>
> **Log [...]

Thanks for the infos.
Your toolchain looks like a native 64-bit compiler, are you sure that your MinGW-toolchain is dual target for the 32bit-architecture? 
The configure log says:

> -- SIL Kit - SILKIT_HOST_PLATFORM Win
> -- SIL Kit - SILKIT_HOST_ARCHITECTURE x86_64
> -- SIL Kit - SILKIT_HOST_COMPILER MinGW

If your MinGW installation has a `i686-w64-mingw32-g++` GCC, you might try that instead of setting the `-m32` C flags.

PRAVEENFrmBOSCH, 2024-09-26:
> Your toolchain looks like a native 64-bit compiler, are you sure that your MinGW-toolchain is dual target for the 32bit-architecture?

Yes, we are building for both 32bit and 64 bit.

> If your MinGW installation has a `i686-w64-mingw32-g++` GCC, you might try that instead of setting the `-m32` C flags.

64 bit also facing the same issue.

MariusBgm, 2024-09-26:
@PRAVEENFrmBOSCH This seems to be a toolchain issue with your mingw setup: I can not reproduce it using my Mingw-W64 installation. Which installation do you use, e.g. is it listed on the official mingw-w64 website? And which kind of mingw build? which runtime is used for the Mingw-build?

For example, I just re-evaluated the SIL Kit build using the Mingw-w64 binaries from winlibs, version 10.5, Win64, for the msvcrt runtime.

Could you provide the output of `cmake --build  --verbose`  -- the verbose output will contain the compiler invocations with all commandline arguments.
In case of the bad images, can you verify that they are valid PE-executables? e.g., invoke the `file` command on the compiled binaries? it should show something like "PE executable". The file command should be part of msys2/mingw/git bash for windows or any WSL distro.

PRAVEENFrmBOSCH, 2024-09-27:
I got it resolved both 32bit and 64 bit works fine , it needs static flags to be enabled.

cmake -G "MinGW Makefiles"  -D CMAKE_C_FLAGS="-m32 -static-libgcc -static-libstdc++ -static" -D CMAKE_CXX_FLAGS="-m32 -static-libgcc -static-libstdc++ -static" ..

MariusBgm, 2024-09-27:
@PRAVEENFrmBOSCH  thanks for reporting back. 
Then it boils down to the missing `libgcc_s.dll, libstdc++.dll` and probably `libwinpthread-1.dll` -- your executables would probably work fine when you run them directly in a msys shell.

I'll close this issue, but feel free to open a new issue if we can be of assistance.

closed 2024-09-27 by MariusBgm