vectorgrp/sil-kit #117: Support for Mac OS

raychenv, 2024-09-20:
Do you have plan to support Mac OS? so development and test of silkit can also be made on Mac OS.

I tried to compile sil-kit (v4.0.52) on Mac OS.
It works. But I need fix some compilation issues in files:
- SilKit/source/util/ExecutionEnvironment.cpp
- SilKit/source/util/SignalHandler.cpp

I ran SilKitDemoCan binary generated, the result is as expected:

> [2024-09-19 13:37:13.299] [CanReader] [info] >> CAN frame: canId=3 timestamp=0s "CAN 0"
> now=0.005s, duration=0.005s
> [2024-09-19 13:37:14.305] [CanReader] [info] >> CAN frame: canId=3 timestamp=0.005s "CAN 1"
> now=0.01s, duration=0.005s
> .....
> [2024-09-19 13:37:13.299] [CanReader] [info] >> CAN frame: canId=3 timestamp=0s "CAN 0"
> now=0.005s, duration=0.005s
> [2024-09-19 13:37:14.305] [CanReader] [info] >> CAN frame: canId=3 timestamp=0.005s "CAN 1"
> now=0.01s, duration=0.005s

MariusBgm, 2024-09-20:
Hi @raychenv , thanks for reaching out and notifying us about the state of macOs.
We definitely support macOs, but it's currently not part of the continuous testing environment. We manually test it, but the latest changes in the files you mentioned uses non-portable linux-isms.

I created an internal issue to track and improve this.

raychenv, 2024-09-20:
@MariusBgm, thanks for the clarification and for opening an internal issue to track this. Looking forward to the improvements.

closed 2024-09-27 by raychenv