vectorgrp/sil-kit #121: Simulation with unknown number of participants

kocho1984, 2024-09-25:
If sil-kit-system-controller has to know number and names of participants in advance, then how can we achieve a simulation where participiants are added and removed dynamically in the runtime?

closed 2024-09-26 by kocho1984

KonradBkd, 2024-09-26:
Hello,

the _sil-kit-system-controller_ is needed if you have a set of participants that need to be available right from the start of the simulation. In SIL Kit terminology, we say that these participants are _required_ and want to perform a _coordinated_ simulation start. This can be achieved by creating the `LifecycleService` with the flag `OperationMode::Coordinated` on these participants. If you look at a single coordinated participant, it doesn't know the about the other participant names that are also _coordinated_. That's were the _sil-kit-system-controller_ comes into play and centrally defines all participant that are required for the simulation to start.

If you want a participant to dynamically join and leave the simulation, we say it runs _autonomously_. In this case, you create a `LifecycleService` with the flag `OperationMode::Autonomous` on these participant, and their names no longer have to appear in the list of names passed to the _sil-kit-system-controller_.

These two concepts can be mixed to be able to cover all kinds of scenarios with the SIL Kit. If you end up with a setup where no participant is required any more and all are dynamic joining and leaving the simulation, just use `OperationMode::Autonomous` for all of them and the _sil-kit-system-controller_ is no longer needed.

Hope this helped!

KonradBkd, 2024-09-26:
Ah seems like you could resolve the issue by yourself just while I was typing, perfect :)

kocho1984, 2024-09-26:
Yes, I solved the problem just by running participants with `--async` flag. In this case simulation works without _sil-kit-system-controller_.

Thank you!

KonradBkd, 2024-09-26:
You're welcome! That's right, the demos have a `--async` flag that instructs them to use the `OperationMode::Autonomous` under the hood. One more feature of the SIL Kit is that this dynamic mode also works if the simulation uses a virtual time. See https://vectorgrp.github.io/sil-kit-docs/simulation/simulation.html#joining-a-running-simulation-with-virtual-time for more info.

kocho1984, 2024-09-26:
@KonradBkd, thanks again for quick and great support!