vectorgrp/sil-kit #129: Removing subscriber/publisher

kocho1984, 2024-10-07:
Is there a way to delete subscriber/publisher which were created with CreateDataSubscriber/CreateDataPublisher methods?

Let's assume that I don't want to be subscibed anymore. How to unsubscibe?
Can I just delete them with delete operator?

VDanielEdwards, 2024-10-07:
Hello @kocho1984, there is no way to delete any 'controller' in SIL Kit after it has been created with any of the `Create...` methods on the `IParticipant` interface.

You can not unsubscribe when you use a publisher/subscriber pair.

I hope this helps!

kocho1984, 2024-10-07:
Hi @VDanielEdwards,

I appreciate you quick response.

OK, ust to make sure: once subscribed, there is no way to stop receiving messages on already subscribed topic?

I also couldn't find any method to remove a participant. So, is the only way to stop receiving messages on a subscribed topic by calling Stop from ILifecycleService?

VDanielEdwards, 2024-10-07:
Yes that is correct, there is no way to stop receiving messages when a publisher and subscriber in the same simulation match (topic, media type, and labels).

A participant is fully removed (and disconnected) from the system if you destroy the `IParticipant` object, which is initially returned as a `std::unique_ptr<IParticipant>` from the `CreateParticipant` function. If you use a a lifecycle it is a good idea to call the `ILifecycleService::Stop` method, prior to destroying the `IParticipant`.

Please note that 'autonomous' participants (i.e., using `OperationMode::Autonomous`) can leave and re-joing a simulation.

closed 2024-10-07 by kocho1984