vectorgrp/sil-kit #130: registryUri in participant configuration file

kocho1984, 2024-10-09:
Hi,

Is there a possibility to put registrUri in configuration file for the participant?

kocho1984, 2024-10-25:
@MariusBgm, is there any easy way to get IP and listening port of sil-kit-registry from IParticipantConfiguration?
I have a config file with following entry and I would like to check in the code using SilKit API what IP and port is in the config.

Middleware:
  RegistryUri: silkit://55.55.55.55:8500

VDanielEdwards, 2024-10-25:
Hello @kocho1984, no this is not possible. The configuration itself is a 'black box' for the application that loads / or uses it.

VDanielEdwards, 2024-10-25:
If you don't mind me asking, what do you need the registry URI for?

kocho1984, 2024-10-25:
I would like to check if IP is available and listening on that port. I have to do that before creating a participant.

VDanielEdwards, 2024-10-25:
The connection to the registry is attempted during the call to `CreateParticipant`. If the connection fails, eg., if it runs into a timeout, or whatever is listening on the port isn't 'speaking' the SIL KIt network protocol like a HTTP server, an exception is thrown which you could catch.

VDanielEdwards, 2024-10-25:
Additionally, SIL Kit will not just try to connect via TCP. At the moment we also attempt to connect via local-domain sockets using a 'predictable' name. But we might add further transports in the future, in addition to the existing ones.

closed 2024-10-25 by kocho1984