vectorgrp/sil-kit #131: Log file timestamp granularity

kocho1984, 2024-10-10:
Hi,

Could you please consider change the format of a timestamp which is added to log file? Currently, it has a second granularity.

What will happen if two participants that use rge same config file are created at the same second? Will they share the same log file? I think that miliseconds granularity would be safer in such cases.

VDanielEdwards, 2024-10-11:
Hello @kocho1984, thank you for posting this issue.

This is an issue that we have faced before as well, if multiple participants are started in the same process in quick succession. We've recently been discussing this internally.

MariusBgm, 2024-10-11:
@kocho1984  @VDanielEdwards  I just merged something that should fix this issue temporarily: https://github.com/vectorgrp/sil-kit/pull/132 
We now add the participant's name to the log file's name.

VDanielEdwards, 2024-10-11:
@kocho1984 and PR #132 has just been merged.

closed 2024-10-16 by kocho1984