vectorgrp/sil-kit #139: Logger lifecycle

kocho1984, 2024-10-25:
1. Is it possible to log to a file with `ILogger` before a participant is created?
Use case: Attempting to create a participant throws an exception and the user wants to log this to a file.

2. Can the logger returned by `participant->GetLogger()` still be used after the participant has been destroyed?
Use case: User wants to log to a file that participant has been destroyed.

VDanielEdwards, 2024-10-28:
Hello @kocho1984, thank you for your questions!

1. Creation of a participant is required to get access to the logger object and there is no way to create logger object otherwise.
2. The logger cannot be used after the participant it belongs to has been destroyed and there is no way to extend it's lifetime beyond that.

However, I do see that both, logging before participant creation, and after participant destruction may be an interesting features for users. I'll create internal issues to track these use-cases and we'll discuss these internally. I cannot promise any outcome though.

Thank you for your patience 👍

kocho1984, 2024-10-28:
@VDanielEdwards: In case `createParticipant` throws an exception, I think logging it to a file would be useful. Currently, logger cannot exist without a participant.

Thank you for your consideration.

closed 2024-12-11 by kocho1984