vectorgrp/sil-kit #156: Attempt to send targeted message to participant, which is not a valid remote receiver

kocho1984, 2024-12-04:
Hello,

What coud be the reason of following error?
```
SilKit-IOWorker: Something went wrong: Error: Attempt to send targeted message to participant 'GVM_participant_173331674198070098bj', which is not a valid remote receiver.
```
I have two participants ECU (subscriber) and GVM_participant_173331674198070098bj (publisher). GVM_participant_173331674198070098bj is trying to sent a message to ECU and it looks like message was sent to SilKit correctly because `Publish()` method didn't throw an exception.

Here is a sequence:
```
[2024-12-04 13:51:10.555] [SystemController] [info] Participant 'ECU' joined the simulation
[2024-12-04 13:51:12.461] [SystemController] [info] Simulation is now running
[2024-12-04 13:52:20.830] [SystemController] [info] Participant 'GVM_participant_173331674198070098bj' joined the simulation
```

GVM_participant_173331674198070098bj sends message to ECU participant using 
`mPublisher[topic]->Publish(serializer.ReleaseBuffer());`

ECU participant doesn't receive that message and following errors appear.

```
[2024-12-04 13:52:23.006] [SystemController] [error] Participant 'GVM_participant_173331674198070098bj' has disconnected without gracefully shutting down.
[2024-12-04 13:52:23.007] [SystemController] [error] SilKit-IOWorker: Something went wrong: Error: Attempt to send targeted message to participant 'GVM_participant_173331674198070098bj', which is not a valid remote receiver.
```

VDanielEdwards, 2024-12-04:
Hello @kocho1984, thank you for creating the issue!

Could you provide us with a [trace log](https://vectorgrp.github.io/sil-kit-docs/configuration/logging-configuration.html) of _both_ participants? This will help us diagnosing the cause of the error.

You can send it to us via e-mail (`silkit` at `vector` dot `com`) if the logs contain information you aren't willing to share publically.

kocho1984, 2024-12-05:
Hi @VDanielEdwards,
I have sent trace logs to address you mentioned.

VDanielEdwards, 2024-12-05:
Thank you! We have received the logs and are investigating.

kocho1984, 2024-12-06:
Hello @VDanielEdwards,
1. Regarding the issue when ECU participant didn't receive message sent by GVM_participant (and no exception was thrown on GVM_participant side), I found out that it happens in case when `Publish()` is called directly after `CreateDataPublisher()`, like in the code below:
 ```C++
 if (mPublisher.count(topic) == 0)
 {
     const std::string mediaType{SilKit::Util::SerDes::MediaTypeData()};
     PubSubSpec dataSpecPubPower{topic, mediaType};
     mPublisher[topic] = mParticipant->CreateDataPublisher("Publisher" + topic, dataSpecPubPower, 0);
 }

mPublisher[topic]->Publish(serializer.ReleaseBuffer());
```
It looks like publisher is not fully ready right away after calling of `CreateDataPublisher()` and data sent by `Publish()` is lost silently and doesn't reach subscriber (ECU participant).

When I added some delay (just for testing), message was received by ECU participant.
 ```C++
 if (mPublisher.count(topic) == 0)
 {
     const std::string mediaType{SilKit::Util::SerDes::MediaTypeData()};
     PubSubSpec dataSpecPubPower{topic, mediaType};
     mPublisher[topic] = mParticipant->CreateDataPublisher("Publisher" + topic, dataSpecPubPower, 0);
     std::this_thread::sleep_for(std::chrono::milliseconds(1000));
 }

mPublisher[topic]->Publish(serializer.ReleaseBuffer());
```
2. Regarding `Attempt to send targeted message to participant, which is not a valid remote receiver` error, I can see that it doesn't happen always. It seems to be not related to problem with not receiving message by ECU because I saw it even if message was received by ECU participant. It happened also in case graceful shutdown of GVM_participant, e.g.
```
[2024-12-06 13:41:50.202] [ECU] [info] Participant 'GVM_participant_17334889102498862ngt' has disconnected after gracefully shutting down
[2024-12-06 13:41:50.203] [ECU] [error] SilKit-IOWorker: Something went wrong: Error: Attempt to send targeted message to participant 'GVM_participant_17334889102498862ngt', which is not a valid remote receiver.
```

VDanielEdwards, 2024-12-09:
Hello @kocho1984, sorry that this took so long - some internal stuff got in the way. I just had a look at the logs and your follow-up post.

As you have figured out yourself, the publish is happening too soon after creating the publisher object.

The [communication-ready handler](https://vectorgrp.github.io/sil-kit-docs/api/services/lifecycle.html?highlight=communicationready#_CPPv4N6SilKit8Services13Orchestration17ILifecycleService28SetCommunicationReadyHandlerE25CommunicationReadyHandler) of the lifecycle service exists for exactly this purpose.
You must wait with any publishing until the callback you registered has been invoked.

Additionally, make sure to create the publisher (and any other controllers) before calling `StartLifecycle`.

I'll see if the documentation needs a more explicit 'note' about creation of controllers and such. Additionally, an error should be logged in this case, the participant should not just silently shut down.

kocho1984, 2024-12-09:
Hello @VDanielEdwards,

> Additionally, make sure to create the publisher (and any other controllers) before calling `StartLifecycle`.
If so, then I'm wondering how to handle a scenario when topic for publisher or subscriber is not known before calling `StartLifecycle`? 

Let's imagine that we have an application which wants to publish data with dynamically created topic every one minute. In such case, each minute, it has to create new publisher with newly generated topic. Does it mean that SilKit doesn't support such a scenario?

VDanielEdwards, 2024-12-09:
Hello @kocho1984, is the 'GVM' participant destroyed immediately after publishing the message? This would explain the error message in the `ECU` participant logs. The 'GVM' participant seems to not shut down normally. Are you waiting for completion of the lifecycle (i.e., calling the `.wait()` or `.get()` methods on the `std::future` returned by `StartLifecycle()`)?

kocho1984, 2024-12-09:
Helo @VDanielEdwards,

GVM participant was shout down properly and that error occured anyway, as you can see in the log attached in point 2 in 
https://github.com/vectorgrp/sil-kit/issues/156#issuecomment-2523600189
```
[2024-12-06 13:41:50.202] [ECU] [info] Participant 'GVM_participant_17334889102498862ngt' has disconnected after gracefully shutting down
[2024-12-06 13:41:50.203] [ECU] [error] SilKit-IOWorker: Something went wrong: Error: Attempt to send targeted message to participant 'GVM_participant_17334889102498862ngt', which is not a valid remote receiver.
```

kocho1984, 2024-12-10:
Hi @VDanielEdwards,

> Additionally, make sure to create the publisher (and any other controllers) before calling StartLifecycle.

Could you please confirm that SilKit doesn't support the dynamic creation of publishers and subscribers after `StartLifecycle()` is called?

I couldn't find any indication in the documentation that this is not supported. However, if it is indeed unsupported, this seems to be a significant limitation which needs to be taken into account when creating architecture of a component which uses SilKit, as all publishers, subscribers, and their topics must be predefined and created before invoking `StartLifecycle()`.

The component I am currently implementing assumes that subscribers and publishers can be created on demand (after `CreateLifecycle()`). This approach generally works as expected, with one exception: when `Publish()` is called immediately after `CreateDataPublisher()`, first published message is silently lost. Therefore, it is crucial for me to confirm whether this approach is unsupported and whether I need to redesign the current architecture.

kocho1984, 2024-12-11:
Problem with `Attempt to send targeted message to participant, which is not a valid remote receiver` error was fixed by calling `wait()` on the `std::future` object returned by `StartLifecycle()`.

@VDanielEdwards, thank you!

closed 2024-12-11 by kocho1984