vectorgrp/sil-kit #185: Working of data transfer

saanjh-sengupta, 2025-03-04:
I wanted to understand how two SIL Kit participants transfer data. 

For instance I have a PUB/SUB based SIL Kit co-simulation b/w 2 participants. I understand that the data transfer is via the Topic mentioned using the  SilKit::Services::PubSub::PubSubSpec &dataSpec

But these participants do not have a NEP attached to it. It is only connected to the registry so how does this data transfer happen ? 

And I can't see this data transfer logs via Wireshark even when the SIL Kit registry IP is filtered in it. 

Could anyone assist on this ?

VDanielEdwards, 2025-03-04:
Hello @saanjh-sengupta, thank you for your question.

As stated in the [documentation](https://vectorgrp.github.io/sil-kit-docs/#guiding-principles) SIL Kit (preferably) establishes peer-to-peer connections between all participants.

The registry acts as a 'broker' for these connections, and provides each participant with the information required to establish the peer-to-peer connections. Only in cases where the peer-to-peer connections cannot be established (e.g., if a firewall blocks these) will the registry act as a 'proxy' for other participants.

However, since SIL Kit (by-default) also tries to use local-domain sockets (aka. UNIX domain sockets), you might not see these connections in Wireshark, even if the participants are connected peer-to-peer.

You can disable usage of local-domain sockets in the SIL Kit participant configuration using the `EnableDomainSockets` flag in the `Middleware` block:

```yaml
Middleware:
  EnableDomainSockets: false
```

saanjh-sengupta, 2025-03-04:
Hi @VDanielEdwards 

Thank you for your response. 

If I have two machines which are co-simulating in this context the SIL Kit registry is in PROXY so that both the participants present in two different machines are connected to SIL Kit. 

In this context, my data gets transferred from 1 Machine to another. 

Is it correct to say that since the Registry is in PROXY - the data transfer is happening via the registry (since it is the only connection point in this context). 

Please clarify.

VDanielEdwards, 2025-03-04:
If both machines cannot reach each other directly - e.g., because their firewalls are blocking all incoming connections - then the registry will be used to proxy the communication between these participants.

We don't encourage using the registry like this, but in some cases it allows starting to work before the lab network has been setup properly. The _proper_ way to use SIL Kit only uses peer-to-peer connections between all participants.

If you want to explicitly forbid a participant from using the registry as a proxy, then you can opt-out using the `RegistryAsFallbackProxy` flag in the participant configuration:

```yaml
Middleware:
  RegistryAsFallbackProxy: false
```

saanjh-sengupta, 2025-03-04:
Hi, 

Thanks for the clarification. 

This answers my query.

closed 2025-03-04 by saanjh-sengupta