vectorgrp/sil-kit #187: How to trigger SendFrame() api for CAN relevant participants for a cyclic interval

AmanChaturvedi24, 2025-03-05:
I have a co-simulation where participnats are communicating over CAN api (OperationMode :: Coordinated)
- Inside the SetSimulationStepHandler i have various function which triggers SendFrame() api's respectively 
- I am hoping to trigger certain of these functions with a particular interval.
For example :
 my StepSize is 10 ms and i want certain functions to be triggered for every 2 ms per simulation cycle 
 So in ideal case, such function should be triggered 5 times per simulation cycle 

Is this something that could be achieved with the current version ?

VDanielEdwards, 2025-03-06:
Hello @AmanChaturvedi24, thank you for your question!

Im not 100% sure that I understand your issue correctly, if the following isn't what you meant, please clarify.

When using SIL Kit with time-synchronization, the participant only sees steps with the duration you specified when registering the simulation step handler. The outgoing messages will be annotated with the timestamp you are informed about in the simulation step handlers `now` argument. There is no way for a SIL Kit participant to produce messages with 'intermediate' timestamps.

If your simulation doesn't care about the timestamps, and that also doesn't care that the frames arrive 'together' (from the point of view of another participant), just send the five frames in the desired order during the simulation step handler.

If the timestamps are important to you, identify the smallest required time step and use that when registering the simulation step handler. You will get more invocations of the simulation step handler, but also get the required timestamps. You will likely have to adjust the logic such that it can deal with these smaller steps.

AmanChaturvedi24, 2025-03-06:
HI @VDanielEdwards,
Thanks for the quick reply,
You got my point as "There is no way for a SIL Kit participant to produce messages with 'intermediate' timestamps." this clears my doubt, 
However regarding the last paragraph i didn't get it fully as  in my case i am having multiple SendFrame Api's sending  different payloads under SetSimulationStepHandler, you can have a look in below picture 

[image]

even if i change my step size to a lesser value lets say 100ms i still did not get how can trigger the SendFrame_3 after every 10 ms so that in one simulation cycle it will get executed 10 times with specific interval in between (i.e. 10ms virtual time).

AmanChaturvedi24, 2025-03-11:
Thank you @VDanielEdwards ,
I got the point, it was really helpful..

closed 2025-03-11 by AmanChaturvedi24