vectorgrp/sil-kit #188: Calling Convention Changes Affecting Multiple Files

TPfitzer, 2025-03-06:
### Description:
Calling convention affect the following files and functions:

**DataSubscriber.hpp** - Method: TheDataMessageHandler
**RpcClient.hpp** - Method: TheRpcCallResultHandler
**RpcServer.hpp** - Method: TheRpcCallHandler
**ThrowOnError.hpp** - Method: ThrowOnError

These methods impact compatibility and require adjustments in the affected components. Further review and potential fixes are needed.

**Steps to Reproduce:**
Change Calling Convention to __stdcall. I assume the compiler will fetch the default calling convention for this methods if no dedicated convention is specified

**Local Solution:**
I fixed the issue locally by adding the Makro SilKitCALL

VDanielEdwards, 2025-03-06:
Thank you for the report (and the fix)!

VDanielEdwards, 2025-03-06:
Just for clarification, what is the issue with the `ThrowOnError` function? It is a free function and is not (?) used as a callback through the C-API.

TPfitzer, 2025-03-06:
@VDanielEdwards you are correct, sorry for the overshoot. Maybe i just changed it in the flow. I just rechecked and its fine without adding calling convention.

VDanielEdwards, 2025-03-06:
@TPfitzer Thank you for the clarification. We'll fix this ASAP.