vectorgrp/sil-kit #192: RPC delay

fskoras, 2025-03-28:
Hello,

I have a requirement to implement RPC calls between participants to set/get some signals and parameters.
I was very happy to see that sil-kit provide back-end implementation for that very purpose.

Please correct me if I'm wrong but current RPC Client/Server works in the following way

1. Request Initiation: An RPC request is made during a participant’s simulation step (tick work time window).
2. Request Registration: The request is queued and scheduled for transmission at the end of the current tick.
3. Request Delivery: The server receives the request in the next tick, as the communication is synchronized with the simulation step boundaries.
4. Server Processing: The server’s request handler processes it and sends a response immediately within the same tick.
5. Response Registration: The response is queued and scheduled for transmission at the end of that tick.
6. Response Delivery: The client receives the response in the subsequent tick.

This results in a minimum delay of two simulation ticks between the initiation of the request and the receipt of the response

Now to my question: Can RPC Request and Response Happen in One Swoop Between Ticks?

I understand that doing things this way for processing communication frames for CAN, FR etc. in co-simulation make sense. But would it be possible to receive RPC response without having to move virtual time by 2 steps? What would be the preferred way to achieve my goal?

Kind Regards
Filip

VDanielEdwards, 2025-03-31:
Hello Filip (@fskoras),

to ensure correct time-stamps and delivery in the expected simulation steps of remote participants, we recommend the process you described – i.e., sending messages only during the next simulation step and not during the reception handler.

When you use the `ITimeSyncService::SetSimulationStepAsync` method, the simulation step is not 'completed' after the callback is returned from, but only after `ITimeSyncService::CompleteSimulationStep` is explicitly called, is the simulation time allowed to progress.

However, this enables you to exchange data _during_ a simulation step – which seems to be you goal.

You will likely need to use a condition variable to signal another thread about the start of the simulation step, make the RPC request, wait for the RPC response. On the corresponding RPC server you will need a similar setup where you react to the reception of the RPC request and trigger sending the response. You can use the async simulation step on both participants.

The [SimStepAsync](https://github.com/vectorgrp/sil-kit/blob/main/Demos/api/Orchestration/SimStepAsync.cpp) demo should provide an idea on how to use the async. simulation step handler and the `ITimeSyncService::CompleteSimulationStep` method.

Hope this helps, if you have further questions on this topic, feel free to post them here.

VDanielEdwards, 2025-03-31:
All good, thanks for the clarification! 😃 

> I assume that the label 'issue next step' marks the time when `CompleteSimulationStep` is called in `Participant2` (and `Participant1`).

This setup should be possible – it does require careful synchronization between the async. simulation step handler and the RPC Server/Client callbacks.

If you don't care about the timestamps, you can send the response in the RPC request callback on `Participant1`.

On `Participant2, I would use a condition variable for wait for the invocation of the RPC response handler.

We don't have and cannot provide blocking operations in SIL Kit.

fskoras, 2025-04-01:
I am using `SetSimulationStepAsync` as you described, but unfortunately, when I make an RPC call and wait for a response, the response never arrives. Additionally, on the server side, the receive callback is not triggered until I use `CompleteSimulationStep`. Therefore, it seems that performing an RPC call and receiving a response within the same simulation step is impossible, even in an async setup. I don’t rule out the possibility that I may be doing something wrong :) but could you please confirm whether this operation is possible?

VDanielEdwards, 2025-04-01:
I've put together a simple example [GH192.zip](https://github.com/user-attachments/files/19553496/GH192.zip) based on the `Demos/api/Orchestration/SimStepAsync.cpp` demo application.

The server side is pretty bare bones and just sends the result as the reversed argument data.
It doesn't actually do anything but wait for 1s in the simulation step loop.

The client side makes a whole call-and-wait-for-result round-trip over a RPC client.

The example does not handle any errors and is also not "pretty", but hopefully demonstrates the idea.

> FYI: I noticed that the demos source code in `Demos/api/Orchestration/SimStepAsync.cpp` has a bug for which I'll push a fix tomorrow.

fskoras, 2025-04-14:
Sorry for the delayed response. It was a silly mistake on my end. Just to sum things up: in order to immediately receive the response, the main thread must be unblocked due to some internal backend logic. So, to achieve this, we need to create a separate thread and allow SetSimulationStepAsync to return before making the actual RPC call. Is that correct?

VDanielEdwards, 2025-04-22:
> Sorry for the delayed response. It was a silly mistake on my end. Just to sum things up: in order to immediately receive the response, the main thread must be unblocked due to some internal backend logic. So, to achieve this, we need to create a separate thread and allow SetSimulationStepAsync callback to return before making the actual RPC call. Is that correct?

No worries! The key is that you must not wait for a response inside of a callback from SIL Kit – e.g., the `SetSimulationStepAsync` callback. How you accomplish this is up to you, but spawning a separate thread and handling the call there is one way to deal with it.

closed 2025-05-06 by fskoras