vectorgrp/sil-kit #214: CAN silkit code crashing intermediately

komalpatil1811, 2025-05-28:
**CAN silkit code is getting aborted intermediately after transmitting few frames.**

Silkit Version: 4.0.55

Source code details: Created thread running at every 1ms and called transmit frames at every 1ms task rate. At receiving end registered reception callback for receiving frames. CAN echo device utility used for testing.

Observed that after sometime the code gets aborted abruptly after Sendframe.

OS: WSL on Windows 11 (x86 gcc make environments)

**Below is code snippet for same:**
#define _GNU_SOURCE
#include <stdio.h>
#include <signal.h>
#include <unistd.h>
#include <string.h>
#include <stdlib.h>
#include <time.h>
#include <errno.h>
#include <sched.h>
#include <pthread.h>
#include <sys/mman.h>
#include <stdatomic.h>
#include "Can.h"
 
#define SIGNAL_NUM (SIGRTMIN)
#define TIMER_INTERVAL_NS 1000000L // 1 ms

volatile sig_atomic_t in_handler = 0;
static timer_t timerid;
static sigset_t sigset_new;
pthread_t thread1;

void GetTimeStamp(char *buffer, size_t buffer_size) 
{
    struct timeval tv;
    struct tm *tm_info;
    gettimeofday(&tv, NULL);
    tm_info = localtime(&tv.tv_sec);
    strftime(buffer, buffer_size, "%Y-%m-%d %H:%M:%S", tm_info);
    // Append microseconds
    size_t len = strlen(buffer);
    snprintf(buffer + len, buffer_size - len, ".%03ld", tv.tv_usec / 1000);
}

void* signal_handler_thread(void* arg) 
{
    Can_PduType g_canPdus;
    uint8_t txHTH;
    uint8_t resultData[8] = {'v', 'C', 'A', 'N', '_', 'D', 'r', 'v'}; // Example data to be sent  
    static int occurence = 0;
    char timestamp[64];

    struct sched_param param = { .sched_priority =  99};
    pthread_setschedparam(pthread_self(), SCHED_FIFO, &param);
    siginfo_t info;
 
    while (1) 
    {
        int sig = sigwaitinfo(&sigset_new, &info);
        if (sig == SIGNAL_NUM) 
        {
            g_canPdus.id = occurence; 
            g_canPdus.length = 0x08; 
            g_canPdus.sdu = resultData; 
            g_canPdus.swPduHandle = 0x10; 
            txHTH = occurence+ 10;
            GetTimeStamp(timestamp, sizeof(timestamp));
            printf("[%s] Can Write Called\n", timestamp);
            Can_Write(txHTH, &g_canPdus);
            GetTimeStamp(timestamp, sizeof(timestamp));
            printf("[%s] Can Write End\n", timestamp);
            occurence++;
            if(occurence>6)
            {
                occurence = 0;
            }
         }
    }
    return NULL;
}

void tpl_start_auto_timer_( long unsigned)
{
    sigemptyset(&sigset_new);
    sigaddset(&sigset_new, SIGNAL_NUM);
    if (pthread_sigmask(SIG_BLOCK, &sigset_new, NULL) != 0)
    {
        perror("pthread_sigmask");
        exit(1);
    }
    // Start signal handler thread
    pthread_create(&thread1, NULL, signal_handler_thread, NULL);
    struct sigevent sev = 
    {
        .sigev_notify = SIGEV_SIGNAL,
        .sigev_signo = SIGNAL_NUM
    };
    if (timer_create(CLOCK_MONOTONIC, &sev, &timerid) != 0) 
    {
        perror("timer_create");
        exit(1);
    }
    struct itimerspec its = 
    {
        .it_value = { .tv_sec = 0, .tv_nsec = TIMER_INTERVAL_NS * 30 },
        .it_interval = { .tv_sec = 0, .tv_nsec = TIMER_INTERVAL_NS }
    };
    if (timer_settime(timerid, 0, &its, NULL) != 0) 
    {
        perror("timer_settime");
        exit(1);
    }
}

int main() 
{
    // Lock memory to avoid page faults during RT
    if (mlockall(MCL_CURRENT | MCL_FUTURE) == -1) 
    {
        perror("mlockall");
    }
    tpl_start_auto_timer_(0);    
    Can_Init();
    for (int i = 0; i < 7; ++i)
    {
        Can_SetControllerMode(i, CAN_CS_STARTED);
    }
    while (1) 
    {
        pause(); // Wait for signals
    }
    return 0;
}

VDanielEdwards, 2025-06-06:
Hello @komalpatil1811,

I'm very sorry for the long delay.

Could you please share a _full_ and _compilable_ example that reproduces the issue you observe? As such we have **zero** information we can use for diagnosing the issue.

Things I noticed:

- `mlockall(MCL_CURRENT | MCL_FUTURE)`
  - this causes all existing and future allocated pages to be backed by physical RAM
  - if you hit `RLIMIT_MLOCK` the process will be termianted via `SIGSEGV` sent by the kernel
  - keep in mind that **all** allocations in the process will be 'automatically' locked in memory, including any made from libraries such as SIL Kit

- adapting the sample code to make it compile- and executable
  - removed all calls to undefined functions and data types
  - added creating a participant and a CAN controller
  - added sending a frame from the timer
  - **could not reproduce a crash**

In short, please provide more information on the crash and a working reproducer.

Thank you and with kind regards
Daniel

komalpatil1811, 2025-06-17:
Hello @VDanielEdwards,

Please find attached Project containing all files.

**It is observed that every time the code is crashed/aborted at the can_write(which internally calls sendframe to silkit).**

**Steps:**
- Standalone_ECU is the executable file

- the project is compilable on x86 environment
- Common folder consists of main() application code
- vMcal folder has all vCAN driver files
- Makefile for build

VDanielEdwards, 2025-06-17:
@komalpatil1811 thank you for the code drop! I was able to compile it and am now trying to reproduce the crash.

komalpatil1811, 2025-06-18:
Hello @VDanielEdwards ,

The results observed are correct. But if we retry again the execution starts properly once CAN controllers are setup correctly.
This will be rectified at my end, but currently i am facing issue once when the transmission is started successfully.
The code crashes within few mins or sometimes runs for few hours and crashes.

I was not able to capture logs, Trying to capture same and once done will share.
Request you to keep the execution running for some hours to check for failure.

VDanielEdwards, 2025-06-24:
Hello @komalpatil1811, I'm currently looking into your latest code drop. Thank you!

VDanielEdwards, 2025-06-24:
@komalpatil1811 I debugged your code with GDB. Put a breakpoint on the `printf("UART Task Terminated\r\n")` line. Then GDB presented me with the following stacktrace:

```
#0  OsTask_uart_1ms_function () at /mnt/d/dev/user/issues/GH214/2025-06-24/Appl/periodic/periodic.c:271
#1  0x0000555555563f6a in tpl_osek_func_stub (task_id=1 '\001') at /mnt/d/dev/user/issues/GH214/2025-06-24/vOS/machines/posix/src/tpl_machine_posix.c:171
#2  0x000055555556444a in tpl_create_context_boot () at /mnt/d/dev/user/issues/GH214/2025-06-24/vOS/machines/posix/src/tpl_posix_context.c:71
#3  0x0000555555564499 in tpl_create_context_trampoline (sigid=10) at /mnt/d/dev/user/issues/GH214/2025-06-24/vOS/machines/posix/src/tpl_posix_context.c:95
#4  <signal handler called>
#5  0x00007ffff72455b9 in __GI___sigsuspend (set=0x7fffffffd3f0) at ../sysdeps/unix/sysv/linux/sigsuspend.c:26
#6  0x0000000000000000 in ?? ()
```

The OS emulation you are using for this project is utilizing signals to switch between the individual tasks. Please note that SIL Kit functions, `printf`, and many more from the C and C++ standard libraries are generally **not** reentrant or signal-safe.

We do not guarantee that SIL Kit functions behave correctly in this setup.

You will have to invoke any SIL Kit functions from a 'normal' thread context and use synchronization primitives that are signal-safe to communicate with this context.

komalpatil1811, 2025-07-31:
Hello @VDanielEdwards : Any further updates.

MariusBgm, 2025-07-31:
@komalpatil1811 I'm not sure I'm getting your comments right.
 We do not do any signal handling in SIL Kit - this is all in your user application and you must adhere to signal safety programming rules. 

Signals are handled in a very limited environment, making it impossible to call into C++ code and C code that is not async-signal-safe. Please refer to the POSIX documentation or your platform docs for this. 
For example, as a starter  see https://man.netbsd.org/signal.7 and "Signal safe functions" in https://man.netbsd.org/sigaction.2.

VDanielEdwards, 2025-10-30:
I'm closing this issue due to inactivity. Please feel free to re-open this issue, or create a new one.